loboris/ESP32_TFT_library #3: Adafruit HUZZAH32 and TFT FeatherWing with ESP-IDF

sukeshak, 2017-09-02:
Trying to get Adafruit HUZZAH32 and Adafruit TFT FeatherWing work with this library.

Updated tftspi.h with the following
#define PIN_NUM_MISO 19		// SPI MISO
#define PIN_NUM_MOSI 18		// SPI MOSI
#define PIN_NUM_CLK  5		// SPI CLOCK pin

#define PIN_NUM_CS   15		// Display CS pin
#define PIN_NUM_DC   33		// Display command/data pin
#define PIN_NUM_TCS  32		// Touch screen CS pin
#define PIN_NUM_RST  16	// GPIO used for RESET control

But still not working...

I also found this setting... not sure if its correct or needs a change
// Define which spi bus to use VSPI_HOST or HSPI_HOST
#define SPI_BUS VSPI_HOST

Any help to get this library working is highly appreciated.

loboris, 2017-09-28:
The pins are correct, so it should work.
I don't think RESET is nessary on Adafruit TFT FeatherWing, try without it...
_#define SPI_BUS VSPI_HOST_ & _#define SPI_BUS HSPI_HOST_ should both work, try both...
Touch cannot be used, the STMPE610 touch controller this board uses is not supported (I'll try to implement it next month). Disable touch.
Try the latest commit...
I don't have Adafruit TFT FeatherWing to try...

sukeshak, 2017-09-28:
You mean set RESET to 0?
I have tried both VSPI & HSPI without any difference (nothing shows up on display).

The SPI sample from ESP-IDF tree works since that implements driver for same display. So not sure what is different between your SPI related code and the one on ESP-IDF source tree.

sukeshak, 2017-09-28:
Link to the discussion on the forum
https://www.esp32.com/viewtopic.php?f=13&t=1683&start=70

loboris, 2017-10-08:
Fixed in the latest commit.

closed 2017-10-08 by loboris