loboris/ESP32_TFT_library #5: font_rotate = 0 issue.

juphaono, 2017-09-21:
Hi loboris,
I just got brand new WROVER_KIT V3 with LCD Type ST7789V. I have downloaded your library and everything works fine. I can see all the demos excepted the text print on the LCD.

I have found that when font_rotate = 0 then it can not print string on the screen.
Then I have tried setting font_rotate = 1 then the strings now being printed.

By the way, thank you for contributing the library.
//juphaono
[image]

loboris, 2017-09-28:
Could you, please, try with the latest commit and report if you still have this issue.

juphaono, 2017-10-01:
Hi loboris,

Thank for correcting the issue. Now the LCD wor
[image]
ks and displays every demos perfectly.

closed 2017-10-05 by loboris