loboris/ESP32_TFT_library #6: Please provide some Screenshots

DKlay, 2017-09-28:
First I would like to thank you for this library. 

Currently, I am trying to integrate this library into my project (C++) with some success but the text is not displayed on my display and fillScreen and drawRect does not work properly (i think) or maybe my display is broken which i will try to replace on this weekend.
btw. i dont used the SPIFFS image.

If possible could expand the project with some images that show the expected drawing results of your demo code?

My Hardware: ESP-WROOM32 with ILI9341 Display.

loboris, 2017-09-28:
I'll try to make a demo application video and put it on YouTube when I'll find some free time...

colesnicov, 2018-03-17:
??