loboris/ESP32_TFT_library #13: Feature requests

DKlay, 2017-10-18:
Hallo Lovosevic,

could you expand the TFT_print function so that it is possible to display the desired text in a certain text size?

Furthermore, I would welcome a function that allows you to draw a line with a certain thickness (irrespective of the angle).
TFT_drawLineByAngle (x, y, start, len, angle, color, thickness = 1 /*as default*/);

csann, 2017-10-20:
Regarding the TFT_print request....won’t this be difficult to do? If I understand how fonts are saved, the embedded fonts are not scaleable. Each font is fixed at a particular size. So to change sizes, you need to change fonts using TFT_setFont. I think file based fonts operate the same way, but I am not sure.

If I am wrong about this, please help me understand how to change font sizes on and I will make this enhancement.

ramonschepers, 2017-10-20:
Not to interrupt this conversation, but here is how the ttf font type works: http://scripts.sil.org/cms/scripts/page.php?site_id=nrsi&id=iws-chapter08

So basically drawing paths which are scaled for the user.

csann, 2017-10-20:
Ramon, I appreciate your information... but are you sure the fonts in this library TrueType fonts? I’m not sure they are.

ramonschepers, 2017-10-22:
They fonts are not ttf in this library, to actually make use of this, you need a (partial) framebuffer for drawing the shapes of a letter in anti alias.

This requires either quite some extra RAM (for 1 screenbuffer) or much more complex code (partial framebuffer)

It's not impossible to do, but would require lots of time to implement correctly.

After all, I just put my 2 cents on this issue.

DKlay, 2017-10-22:
Hallo,

I can't say much about fonts, but you can look at the drawChar method under the following source code (https://github.com/Bodmer/TFT_ILI9341/blob/master/TFT_ILI9341. cpp).

Don't misunderstand I find the possibility to load and use font files good, but mostly I only want to output a simple text, sometimes only with certain sizes.

loboris, 2017-10-22:
Scaling bitmap fonts usualy produces ugly looking fonts, that is the main reason why font scaling is not implemented.
In this library you can have as many fonts as you want, so you can easily create the same font in multiple sizes and name them for example myFont8, myFont10, myFont16, ...
The compiled font size is usualy 1~3 KB, so it is not an issue to include many fonts as internal font or keep them in files.
Using the [included tool](https://github.com/loboris/ESP32_TFT_library/tree/master/tools) you can convert any ttf font with desired size to c file which can be included as internal font or compiled to file font.