loboris/ESP32_TFT_library #14: Trouble Compiling tft_demo in ESP32_TFT_library

jimwims, 2017-10-22:
I am getting the messages shown below when attempting to compile the demo app to run on my ESP32 Rover board.  Any suggestions?

CC /Users/jim/esp/ESP32_TFT_library/build/spiffs/spiffs_vfs.o
/Users/jim/esp/ESP32_TFT_library/components/spiffs/./spiffs_vfs.c: In function vfs_spiffs_register':
/Users/jim/esp/ESP32_TFT_library/components/spiffs/./spiffs_vfs.c:836:9: error:unknown field 'fd_offset' specified in initializer
         .fd_offset = 0,
         ^
/Users/jim/esp/ESP32_TFT_library/components/spiffs/./spiffs_vfs.c:837:18: warning: initialized field overwritten [-Woverride-init]
         .flags = ESP_VFS_FLAG_DEFAULT,
                  ^
/Users/jim/esp/ESP32_TFT_library/components/spiffs/./spiffs_vfs.c:837:18: note:(near initialization for 'vfs.flags')
make[1]: *** [spiffs_vfs.o] Error 1
make: *** [component-spiffs-build] Error 2

alabig, 2017-10-22:
I am also getting this error, but it worked perfectly a week ago before I moved to a new location and updated everything.. this might be why:

https://github.com/espressif/esp-idf/commit/539262b5c2f186154092d727863dc39cb0c2ceee

Looks like esp-idf was updated very recently to remove the field. I just reverted to a previous version of esp-idf and compiled with no problems (i.e. it is working again).

loboris, 2017-10-25:
For now, just comment the line 836 in components/spiffs/spiffs_vfs.c
https://github.com/loboris/ESP32_TFT_library/blob/015a576adf692c17f316b6746b2e53d169307fb4/components/spiffs/spiffs_vfs.c#L836
The fix will be pushed soon (I'm working on some other changes too).

jimwims, 2017-10-25:
That was easy.  Thanks very much!

loboris, 2017-11-01:
Fixed in latest commit.

closed 2017-11-01 by loboris