loboris/ESP32_TFT_library #18: WiFi Initialization Fails. Doesn't connect.

tkamaal, 2017-11-04:
I'm ESP32_TFT_library on ESP32_Wrover_Kit V3. I was able to display Image on LCD without using WiFi. As a part of next step, I wanted to use WiFi so I followed the steps provided in README and made modifications in using make menuconfig. I have provided WiFi SSID and WiFi Password. 

I get the following error. 

W (2855) wifi : wifi nvs_open fail ret = 4353
E (2857) wifi : wifi_init 1410 ret = 4353
ESP_ERROR_CHECK failed : esp_err_t 0x1101 at 0x4014bde

CPU halted. 

By going through the source code, I wasn't able to figure out what the error code meant. I looked into the following files esp_err.h and esp_wifi.h.

tkamaal, 2017-11-04:
[image]

I have highlighted the error message.

loboris, 2017-11-06:
Sorry for the bug, it is fixed now.

tkamaal, 2017-11-06:
Thank you @loboris

closed 2017-11-06 by tkamaal