loboris/ESP32_TFT_library #19: load_file_font

nottledim, 2017-11-10:
First time I've seen a "goto" in code for a long time.  Guess what - there a bug!

in component/tft/tft.c

If, for some reason the file size check (or one of the other tests) fails after the file is opened it's essential that the file closed otherwise spiffs quickly runs out of file handles.

adding a close(fhndl) after the exit label helps.

loboris, 2017-11-11:
Thank you for reporting this.
The file should be opened **after** the size check. It is closed imediately after reading, so there is no need to close it after exit label.

nottledim, 2017-11-11:
That's a better solution, thanks

nottledim, 2017-11-11:
Whilst you're fixing that would you consider making error numbers reported by compile_font_file and load_file_font unique.  Also perhaps print err as well err_msg in penultimate line of compile_font_file.
Thanks.