loboris/ESP32_TFT_library #22: XPT2046 touch calculation

nottledim, 2017-11-24:
I'm not having much success getting touch to work.  There is one show stopper in  TFT_read_touch in tft.c.

There is a test for (ybottom - ytop) <= 0 which is always true and so calculation is always abandoned.

The constant TP_CALY_XPT2046  = 122224794 === 0x749009A
Masking that value with 0x3FFF returns 0x009A - that's ybottom
Shifting TP_CALY right by 16 bits and masking with 0x3FFF returns 0x0749 - thats ytop.

It's a constant calculation so I'm puzzled that it's recomputed every reading.  I'm not too sure what's intended.

SindhujaNarra, 2018-06-07:
Hi nottledim, 

I'm having the similar issue with the XPT2046 touch calculations. Were you able to resolve it? Please let me know if you have found the solution to this issue. 

Thanks in advance!

nottledim, 2018-06-07:
oh, gosh -  yes and no.  I got it to work well enough for what I wanted by 
fiddling about.  I don't think you could call it a resolution.

If I remember there was a problem with the mask used to select the position 
value from the chip register.

I'll dig the code out tomorrow to remind myself and I'll also send it to you.

Dick

SindhujaNarra, 2018-06-07:
That would be great. Thanks a lot!

SindhujaNarra, 2018-06-11:
Hi nottledim, I am still struggling to get the correct values for X and Y. Could you please help me with the same?

earosenfeld, 2019-07-11:
I am experiencing the same issue, was anyone able to resolve the issue with the XPT2046 touch calculation?
Thank you