loboris/ESP32_TFT_library #24: JPEG and BMP not shown in demo

dmetzinger, 2017-12-08:
JPEG and BMP do not display on ESP-WROVER-KIT. All other functions display correctly.

make clean & make erase_flash -> make all & make flashfs -> make flash monitor

Built with latest esp-idf (as of 12/7/2017) and xtensa (1.22.0-75-gbaf03c2-5.2.0)

> ////MONITOR OUTPUT
> MONITOR
> --- idf_monitor on /dev/ttyUSB1 115200 ---
> --- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0018,len:4
> load:0x3fff001c,len:5728
> ho 0 tail 12 room 4
> load:0x40078000,len:0
> load:0x40078000,len:13816
> entry 0x4007903c
> I (74) boot: Detected ESP32
> I (34) boot: ESP-IDF v3.0-dev-1411-g7e3ac347 2nd stage bootloader
> I (34) boot: compile time 22:00:31
> I (37) boot: Enabling RNG early entropy source...
> I (40) boot: SPI Speed      : 40MHz
> I (44) boot: SPI Mode       : DIO
> I (48) boot: SPI Flash Size : 4MB
> I (52) boot: Partition Table:
> I (56) boot: ## Label            Usage          Type ST Offset   Length
> I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
> I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
> I (78) boot:  2 factory          factory app      00 00 00010000 00100000
> I (85) boot:  3 storage          Unknown data     01 82 00180000 00100000
> I (93) boot: End of partition table
> I (97) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0b510 ( 46352) map                                                                                
> I (122) esp_image: segment 1: paddr=0x0001b538 vaddr=0x3ffb0000 size=0x04ad8 ( 19160) load                                                                              
> I (130) esp_image: segment 2: paddr=0x00020018 vaddr=0x400d0018 size=0x22348 (140104) map                                                                               
> 0x400d0018: _stext at ??:?
> 
> I (179) esp_image: segment 3: paddr=0x00042368 vaddr=0x3ffb4ad8 size=0x005c4 (  1476) load                                                                              
> I (180) esp_image: segment 4: paddr=0x00042934 vaddr=0x40080000 size=0x00400 (  1024) load                                                                              
> 0x40080000: _iram_start at /home/metzingerd/esp/esp-idf/components/freertos/./xtensa_vectors.S:1685                                                                     
> 
> I (186) esp_image: segment 5: paddr=0x00042d3c vaddr=0x40080400 size=0x0a51c ( 42268) load                                                                              
> I (212) esp_image: segment 6: paddr=0x0004d260 vaddr=0x400c0000 size=0x00000 (     0) load                                                                              
> I (219) boot: Loaded app from partition at offset 0x10000
> I (219) boot: Disabling RNG early entropy source...
> I (221) cpu_start: Pro cpu up.
> I (225) cpu_start: Starting app cpu, entry point is 0x40080ea8
> 0x40080ea8: call_start_cpu1 at /home/metzingerd/esp/esp-idf/components/esp32/./cpu_start.c:222                                                                          
> 
> I (0) cpu_start: App cpu up.
> I (236) heap_init: Initializing. RAM available for dynamic allocation:
> I (242) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
> I (248) heap_init: At 3FFB5A08 len 0002A5F8 (169 KiB): DRAM
> I (255) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
> I (261) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
> I (267) heap_init: At 4008A91C len 000156E4 (85 KiB): IRAM
> I (273) cpu_start: Pro cpu start user code
> I (68) cpu_start: Starting scheduler on PRO CPU.
> I (0) cpu_start: Starting scheduler on APP CPU.
> 
> ==============================
> TFT display DEMO, LoBo 11/2017
> ==============================
> Pins used: miso=25, mosi=23, sck=19, cs=22
> ==============================
> 
[... 526 lines cut ...]
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072

#
# SPI Flash driver
#
CONFIG_SPI_FLASH_VERIFY_WRITE=
CONFIG_SPI_FLASH_ENABLE_COUNTERS=
CONFIG_SPI_FLASH_ROM_DRIVER_PATCH=y
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS=
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED=

#
# tcpip adapter
#
CONFIG_IP_LOST_TIMER_INTERVAL=120

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096
```

sergey-kaydalov, 2017-12-27:
Found the same problem. Also noticed user fonts aren't loaded from SPIFFS.

andywang2039, 2018-01-19:
hello everybody, i meet same problem that is no JPEG and BMP display on demo, i used is ESP-WROVER-KIT V3, the LCD just dispaly white lights, even i changed CONFIG_SPIFFS_BASE_ADDR=0x180000 to try, but result is sames. how can i fix it?

janekm, 2018-01-30:
@andywang2039 I had the same problem, eventually figured out that my ESP-WROVER-KIT v3 (bought direct from Espressif...) has a display with ILI9341 driver IC, not ST7789V like it's supposed to. Had me questioning my sanity since that doesn't seem to be documented anywhere, boo! Only noticed after I tried the master-spi demo in the idf which detected the driver.

vibnwis, 2018-02-26:
I face the same problem as well. Mine is M5Stack. Checked the sdkconfig CONFIG_SPIFFS_BASE_ADDR is 180000. Running out of clues. Any advice, please?

vibnwis, 2018-02-26:
I have got it working, but the image demo are still blank despite "make flashfs". I tried make copyfs as well.

vibnwis, 2018-02-26:
Further work, I found out that the jpeg files reading errors. See below. I don't have the solution yet. Feel free to comment, please?

Display: ILI9341: PORTRAIT 240,320 Color

     Clear screen time: 61 ms
Send color buffer time: 229 us (240 pixels)
I (127242) event: station ip lost
File error: Bad file numbers
JPG: filename str /spiffs/images/test1.jpg 
    JPG time, scale: 5: 55 ms
File error: Bad file numbers
JPG: filename str /spiffs/images/test1.jpg 
    JPG time, scale: 4: 54 ms
File error: Bad file numbers
JPG: filename str /spiffs/images/test1.jpg 
    JPG time, scale: 3: 54 ms
File error: Bad file numbers
JPG: filename str /spiffs/images/test1.jpg 
    JPG time, scale: 2: 54 ms
File error: Bad file numbers
JPG: filename str /spiffs/images/test1.jpg 
    JPG time, scale: 1: 55 ms
File error: Bad file numbers
JPG: filename str /spiffs/images/test1.jpg 
    JPG time, scale: 0: 54 ms
File error: Bad file numbers
JPG: filename str /spiffs/images/test2.jpg 
    JPG time, scale: 5: 54 ms
File error: Bad file numbers
JPG: filename str /spiffs/images/test2.jpg 
    JPG time, scale: 4: 54 ms
File error: Bad file numbers
JPG: filename str /spiffs/images/test2.jpg 
    JPG time, scale: 3: 54 ms
File error: Bad file numbers
JPG: filename str /spiffs/images/test2.jpg 
    JPG time, scale: 2: 54 ms
File error: Bad file numbers
JPG: filename str /spiffs/images/test2.jpg 
    JPG time, scale: 1: 54 ms
File error: Bad file numbers
JPG: filename str /spiffs/images/test2.jpg 
    JPG time, scale: 0: 54 ms
File error: Bad file numbers
JPG: filename str /spiffs/images/test4.jpg 
    JPG time, scale: 5: 54 ms
File error: Bad file numbers
JPG: filename str /spiffs/images/test4.jpg 
    JPG time, scale: 4: 54 ms
File error: Bad file numbers
JPG: filename str /spiffs/images/test4.jpg 
    JPG time, scale: 3: 54 ms
File error: Bad file numbers
JPG: filename str /spiffs/images/test4.jpg 
    JPG time, scale: 2: 54 ms
File error: Bad file numbers
JPG: filename str /spiffs/images/test4.jpg 
    JPG time, scale: 1: 54 ms
File error: Bad file numbers
JPG: filename str /spiffs/images/test4.jpg 
    JPG time, scale: 0: 54 ms
Error: -1 [opening file]
BMP: filename str /spiffs/images/tiger.bmp 
    BMP time, scale: 5: 55 ms
Error: -1 [opening file]
BMP: filename str /spiffs/images/tiger.bmp 
    BMP time, scale: 4: 54 ms
Error: -1 [opening file]
BMP: filename str /spiffs/images/tiger.bmp 
    BMP time, scale: 3: 55 ms
Error: -1 [opening file]
BMP: filename str /spiffs/images/tiger.bmp 
    BMP time, scale: 2: 54 ms
Error: -1 [opening file]
BMP: filename str /spiffs/images/tiger.bmp 
    BMP time, scale: 1: 54 ms
Error: -1 [opening file]
BMP: filename str /spiffs/images/tiger.bmp 
    BMP time, scale: 0: 54 ms

vibnwis, 2018-02-26:
"make makefs
to create spiffs image in build directory without flashing to ESP32"

type make makefs, and looked into build/spiffs_image, found out that the spiffs images are created. 
What is spiffs image? can I just copy the *.jpeg files?

vibnwis, 2018-02-26:
make makefs creates spiffs_image.img in build folder, not in build/spiffs_image folder. I tried by copying the spiffs_image.img into build/spiffs_image folder. Tested it does not work.

vibnwis, 2018-02-26:
1. make flashfs caused corruption on user program. If you make flashfs, user program will stop running.
The solution for this is by using https://github.com/loboris/ESP32_spiffs_example to "make flashfs" after users have uploaded their program. User can run ESP32_spiffs_example demo to verify if the images are still intact.
2. The function disp_images() does not show all the images and scale each image at all scale level. I have modified to show all images to do that.
//-------------------------
static void disp_images() {
    uint32_t tstart;
    char * filename = NULL;

	disp_header("JPEG IMAGES");

	if (spiffs_is_mounted) {
		// ** Show scaled (1/8, 1/4, 1/2 size) JPG images
		//TFT_jpg_image(CENTER, CENTER, 3, SPIFFS_BASE_PATH"/images/test1.jpg", NULL, 0);
		//Wait(500);

		for (int scale=5; scale >= 0; scale--) {
					tstart = clock();
					// ** Show scaled (1/8, 1/4, 1/2 size) JPG images
					filename = SPIFFS_BASE_PATH"/images/test1.jpg";
					TFT_jpg_image(CENTER, CENTER, scale,filename, NULL, 0);
					if (doprint) printf("JPG: filename str %s \r\n", filename);
					//TFT_bmp_image(CENTER, CENTER, scale, SPIFFS_BASE_PATH"/images/tiger.bmp", NULL, 0);
					tstart = clock() - tstart;
					if (doprint) {
						printf("    JPG time, scale: %d: %u ms\r\n", scale, tstart);
					}
					sprintf(tmp_buff, "Decode time: %u ms", tstart);
					update_header(NULL, tmp_buff);
					Wait(-500);
		}

		//TFT_jpg_image(CENTER, CENTER, 2, SPIFFS_BASE_PATH"/images/test2.jpg", NULL, 0);
		//Wait(500);

		for (int scale=5; scale >= 0; scale--) {
							tstart = clock();
							// ** Show scaled (1/8, 1/4, 1/2 size) JPG images
							filename = SPIFFS_BASE_PATH"/images/test2.jpg";
							TFT_jpg_image(CENTER, CENTER, scale,filename, NULL, 0);
							if (doprint) printf("JPG: filename str %s \r\n", filename);
							//TFT_jpg_image(CENTER, CENTER, 3, SPIFFS_BASE_PATH"/images/test1.jpg", NULL, 0);
							//TFT_bmp_image(CENTER, CENTER, scale, SPIFFS_BASE_PATH"/images/tiger.bmp", NULL, 0);
							tstart = clock() - tstart;
							if (doprint) printf("    JPG time, scale: %d: %u ms\r\n", scale, tstart);
							sprintf(tmp_buff, "Decode time: %u ms", tstart);
							update_header(NULL, tmp_buff);
							Wait(-500);
				}

		for (int scale=5; scale >= 0; scale--) {
							tstart = clock();
							// ** Show scaled (1/8, 1/4, 1/2 size) JPG images
							filename = SPIFFS_BASE_PATH"/images/test3.jpg";
							TFT_jpg_image(CENTER, CENTER, scale,filename, NULL, 0);
							if (doprint) printf("JPG: filename str %s \r\n", filename);
							//TFT_jpg_image(CENTER, CENTER, 3, SPIFFS_BASE_PATH"/images/test1.jpg", NULL, 0);
							//TFT_bmp_image(CENTER, CENTER, scale, SPIFFS_BASE_PATH"/images/tiger.bmp", NULL, 0);
							tstart = clock() - tstart;
							if (doprint) printf("    JPG time, scale: %d: %u ms\r\n", scale, tstart);
							sprintf(tmp_buff, "Decode time: %u ms", tstart);
							update_header(NULL, tmp_buff);
							Wait(-500);
				}

		//TFT_jpg_image(CENTER, CENTER, 1, SPIFFS_BASE_PATH"/images/test4.jpg", NULL, 0);
		//Wait(500);

		for (int scale=5; scale >= 0; scale--) {
									tstart = clock();
									// ** Show scaled (1/8, 1/4, 1/2 size) JPG images
									filename = SPIFFS_BASE_PATH"/images/test4.jpg";
									TFT_jpg_image(CENTER, CENTER, scale,filename , NULL, 0);
									if (doprint) printf("JPG: filename str %s \r\n", filename);
									//TFT_jpg_image(CENTER, CENTER, 2, SPIFFS_BASE_PATH"/images/test2.jpg", NULL, 0);
									//TFT_jpg_image(CENTER, CENTER, 3, SPIFFS_BASE_PATH"/images/test1.jpg", NULL, 0);
									//TFT_bmp_image(CENTER, CENTER, scale, SPIFFS_BASE_PATH"/images/tiger.bmp", NULL, 0);
									tstart = clock() - tstart;
									if (doprint) printf("    JPG time, scale: %d: %u ms\r\n", scale, tstart);
									sprintf(tmp_buff, "Decode time: %u ms", tstart);
									update_header(NULL, tmp_buff);
									Wait(-500);
						}
#if 0
		// ** Show full size JPG image
		tstart = clock();
		TFT_jpg_image(CENTER, CENTER, 0, SPIFFS_BASE_PATH"/images/test3.jpg", NULL, 0);
		tstart = clock() - tstart;
		if (doprint) printf("       JPG Decode time: %u ms\r\n", tstart);
		sprintf(tmp_buff, "Decode time: %u ms", tstart);
		update_header(NULL, tmp_buff);
		Wait(-GDEMO_INFO_TIME);
#endif
		// ** Show BMP image
		update_header("BMP IMAGE", "");
		for (int scale=5; scale >= 0; scale--) {
			tstart = clock();
			filename = SPIFFS_BASE_PATH"/images/tiger.bmp";
			TFT_bmp_image(CENTER, CENTER, scale, filename, NULL, 0);
			if (doprint) printf("BMP: filename str %s \r\n", filename);
			tstart = clock() - tstart;
			if (doprint) printf("    BMP time, scale: %d: %u ms\r\n", scale, tstart);
			sprintf(tmp_buff, "Decode time: %u ms", tstart);
			update_header(NULL, tmp_buff);
			Wait(-500);
		}
		Wait(-GDEMO_INFO_TIME);
	}
	else if (doprint) printf("  No file system found.\r\n");
}

vibnwis, 2018-02-27:
FYI, I have added the test_spiffs function which I modified from https://github.com/loboris/ESP32_spiffs_example. The intention is to show spiffs partition and its files are intact. If you are interested, email me at siowlimchow@gmail.com.

cjsm74, 2018-03-03:
Under "TFT Display DEMO Configuration" at SPIFFS Base address there is a "0x" missing in front of the 180000 value. It is currently interpreted as a decimal value instead of hex.

rohansingh, 2018-04-19:
@janekm Thanks a ton, had the same issue and would never have figured it out.

colesnicov, 2018-04-22:
Great!

ronanmcall, 2018-04-29:
THANK YOU -- fixed my issue!  

Bought my ESP32-WROVER-KIT a few days ago from Mouser, *hoping* it was a v3 -- but it has an ILI9341 but otherwise it looks (black matte) like a V3 which is supposed to have ST7789V (in v2/v3) so I guess it is a V1 or something else (branded Expressif).  the display OD measured exactly that of the ILI9341.

This solution to get the images working is greatly appreciated!
Ronan

jjfretired, 2018-05-06:
You should also change SPIFFS BASE ADDRESS in main/Kconfig.prebuild to 0x180000.  and range.

ronammcall- The board with the ILI9341 is a V3 not a V1.  Check the documentation and you can verify it.

RemyEE, 2018-06-25:
@360art thanks. It works for me.

Jekyll555, 2018-12-14:
Might it be you are missing flashing the spiffs partition from the image file (provided in this repo /components/spiffs_image/spiffs_image.img).
...
`python ${IDF_PATH}/components/esptool_py/esptool/esptool.py --chip esp32 --port COMxx --baud 921600 write_flash --flash_size detect 0x180000 yourProject/components/spiffs_image/spiffs_image.img`

romocitto88, 2021-01-25:
Hi, I'm facing in the same issue. I'm using Platformio on Windows, can someone help me with the @360art solution reproduced on Platformio?

vibnwis, 2021-01-25:
Have you followed the recommended steps?

romocitto88, 2021-01-26:
Yes, the steps:
- Builded the tft_demo example;
- Flashed the builded example;
- I executed the command "pio run --target buildfs -v"
- I executed the command "pio run --target flashfs -v"

vibnwis, 2021-01-26:
Nope. See it above, https://github.com/loboris/ESP32_TFT_library/issues/24#issuecomment-380583113
from 360art

romocitto88, 2021-01-26:
Great instruction if I knew where to run "make ...". I'm using Windows 10. Can you point my in the right direction please?

vibnwis, 2021-01-26:
Sorry, I am not Windows user. However, it should not be hard. Google it,
you will get the answers.

romocitto88, 2021-01-27:
What is the difference doing:
- Upload from platformio (make build && make flash)
- Exec (make makefs)
> mkspiffs.exe -c [..]\ESP32_TFT_library-master\data -b 4096 -p 256 -s 1114112 image.img
- Exec (make flashfs)
> esptool.py --chip esp32 --port "COM30" --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_size detect 0x110000 [..]\image.img

vibnwis, 2021-01-27:
Often this is not the best place for getting those answers, instead reading
the ESP32 user manual. One good reason is, a blind leading a blind.
However, user manual is skwsys accurate.