loboris/ESP32_TFT_library #33: image not being displayed

jjfretired, 2018-04-04:
Followed instructions and configured menuconfig, compiled and make flash, make  flashfs for wrover kit V3.  Display light but no image displayed.  Then it hanges.

V (4792) tcpip_adapter: check: local, if=0 fn=0x400ec028^M
^M
V (4797) tcpip_adapter: call api in lwip: ret=0x5001, give sem^M
V (4802) tcpip_adapter: check: remote, if=0 fn=0x400ec028^M
^M
V (4808) event: exit default callback^M
^M^M
^M
I (6889) [SPIFFS]: Registering SPIFFS file system^M
I (6889) [SPIFFS]: Mounting SPIFFS files system^M
I (6889) [SPIFFS]: Start address: 0x180000; Size 1024 KB^M
I (6892) [SPIFFS]:   Work buffer: 2048 B^M
I (6895) [SPIFFS]:    FDS buffer: 384 B^M
I (6899) [SPIFFS]:    Cache size: 2048 B^M
I (6974) [SPIFFS]: Mounted^M
^M^M
==========================================^M^M
Display: ST7789V: PORTRAIT 240,320 Color^M^M
^M^M
     Clear screen time: 60 ms^M^M
Send color buffer time: 229 us (240 pixels)^M^M
D (124777) tcpip_adapter: if0 ip lost tmr: enter^M
D (124777) tcpip_adapter: if0 ip lost tmr: raise ip lost event^M
D (124777) event: SYSTEM_EVENT_STA_LOST_IP^M
V (124780) event: enter default callback^M
I (124783) event: station ip lost^M
V (124786) event: exit default callback^M
       JPG Decode time: 85 ms^M^M
    BMP time, scale: 5: 86 ms^M^M
    BMP time, scale: 4: 85 ms^M^M
    BMP time, scale: 3: 85 ms^M^M
    BMP time, scale: 2: 85 ms^M^M
    BMP time, scale: 1: 85 ms^M^M
    BMP time, scale: 0: 85 ms^M^M
^M^M
==========================================^M^M
Display: ST7789V: LANDSCAPE 320,240 Color^M^M
^M^M
     Clear screen time: 57 ms^M^M
Send color buffer time: 302 us (320 pixels)^M^M
ets Jun  8 2016 00:22:57^M
^M
rst:0x1 (POWERON_RESET),boot:0x26 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_FEO_V2))^M
waiting for download^M

jjfretired, 2018-04-07:
Added pulls and it worked but images still not being displayed

Grtschnk, 2018-05-21:
Have you seen #35?
Had the same issue, above mentioned post worked for me.

jjfretired, 2018-05-29:
See issue 40.

closed 2018-05-29 by jjfretired