loboris/ESP32_TFT_library #37: SD card and Display on the same SPI bus

amardhore, 2018-05-04:
I tried using SD write every 30 seconds along with display. I took care of CS select and deselect of the display when writing to the SD. Display seems to freeze and never recovers. Below is my sample code. Did anyone try writing to SD card? Thank you for the help.
`	while (1) {

	Wait(10000);
		if(waitSDwrite>3)
		{
	Wait(1000);
	// ==== Test select/deselect ====

	int ret = spi_lobo_device_deselect(spi);
    assert(ret==ESP_OK);

			waitSDwrite = 0;
			test_sd_card();

	ret = spi_lobo_device_select(spi, 1);
    assert(ret==ESP_OK);
	Wait(1000);

		}
		else 
		{
TFT_fillWindow(TFT_BLACK);
			_fg =  TFT_RED;
TFT_print("Hello BluLeaf\n", CENTER, CENTER);
			waitSDwrite++;
		}
}`

sukeshak, 2018-05-30:
@loboris 
Please let us know how to use both display and SD card at the same time.

No reference on how to use these functions as well
- disp_select() Activate display's CS line
- disp_deselect() Deactivate display's CS line

petestewardson-schneider, 2018-08-10:
Yes please give guidance on using your driver alongside touch controllers and SD cards on SPI at the same time.

an-erd, 2019-01-08:
Any news on this topic? I don't get TFT Display, SD Card and PSRAM, all on SPI bus (using this library for TFT and SD card, and the esp-idf SPI library for PSRAM), running at the same time. Component by component is fine.
Display freezes, so at least a way to recover the display is necessary.

chegewara, 2019-04-01:
Hi,
ive been playing a bit today with display driver and sd card and i found where is the problem (why display is freezing), but i dont know how to fix it (just didnt have time to find fix).

The problem is that spi clock speed after using sd card is changed or other settings, because when
Ive been trying to change back clock with `spi_lobo_set_speed` i stuck on this https://github.com/loboris/ESP32_TFT_library/blob/master/components/spidriver/spi_master_lobo.c#L888, as far as i can tell if should return false, but after using sd card this if is returning true.

chegewara, 2019-04-01:
I found workaround, but its very ugly. After using sd card i am doing this:
```
spi_lobo_bus_remove_device(disp_spi);

start_spi_display(); <--- full display initialization
```

jkearins, 2019-06-13:
Hi, i'm trying to use 2 devices on one bus. TFT as 1rst device works fine. When I call spi_lobo_device_select() for 2nd device the function calls spi_lobo_bus_free(1, 0):
[https://github.com/loboris/ESP32_TFT_library/blob/master/components/spidriver/spi_master_lobo.c#L731](url)
It seems to be an error, instead of '1' it shuld be 'i':
spi_lobo_bus_free(i, 0);
I changed '1' to 'i', but still the function returns an error, so select fails.