loboris/ESP32_TFT_library #42: How to integrate to a project properly?

janoist1, 2018-06-15:
Hi,
Can you help me how to integrate it as a component into an ESP-IDF project? I cloned the lib into the `components` folder but unable to compile the demo app, complaining about missing tft files... If I only merge the components folder into my project I can use it... Not sure which way is the correct one. Thanks

sukeshak, 2018-06-19:
Folders inside the component folder is all you need for integration.
For menuconfig options you just need to copy Kconfig.projbuild from main folder to your main folder.
If you are using SPIFF storage then partitions_example.csv file also need to be copied.

jeremyjh, 2019-09-17:
I've addressed this in my fork (for the latest build system in esp-idf 4). All the menuconfig is moved into the TFT components. You just need to add a couple of lines in your CMake files, its all documented here: https://github.com/jeremyjh/ESP32_TFT_library#installing-as-library