loboris/ESP32_TFT_library #43: CENTER and RIGHT in clipwin

m-byte, 2018-07-06:
When I try to use RIGHT inside a clipwin, right align is still done on the full screen, but any character outside the clipwin is not displayed. CENTER also uses the full screen as reference.
Expected behavior would be, that these functions are used relative to the current clipwin.

tsqdhh, 2018-07-27:
The bug is in line 1957 in tft.c.

if (x == RIGHT) x = dispWin.x2 - tmpw + dispWin.x1;
change:
	if (x == RIGHT) x = dispWin.x2 - tmpw ;

MarkJeronimus, 2021-06-25:
I concluded the same bugged line of code. Apart from disappearing characters, they also don't align correctly to the right of the dispWin.

[image]