loboris/ESP32_TFT_library #48: heap corruption when printing 'g'

rpatel3001, 2018-08-08:
I'm experiencing heap corruption when trying to print a string containing the lowercase letter g using the TFT_print function. The text up to the letter g is printed, and the part of the g that looks like an a, but the descender and the rest of the text is cut off. The esp crashes with 

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.                        
Core 0 register dump:                                                                                     
PC      : 0x4008f6c3  PS      : 0x00060b33  A0      : 0x80082824  A1      : 0x3ffbb0e0                    
0x4008f6c3: is_free at /home/rajan/esp/esp-idf/components/heap/multi_heap.c:377                           
 (inlined by) multi_heap_malloc_impl at /home/rajan/esp/esp-idf/components/heap/multi_heap.c:429          
                                                                                                          
A2      : 0x3ffb9598  A3      : 0x0000014c  A4      : 0x3ffba044  A5      : 0x00000000                    
A6      : 0x00000840  A7      : 0x00000200  A8      : 0x00000001  A9      : 0xff00d044                    
A10     : 0x3ffba044  A11     : 0x00000001  A12     : 0x3ffba048  A13     : 0x3ffb95a8                    
A14     : 0x00000344  A15     : 0x3ffba150  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c                    
EXCVADDR: 0xff00d044  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff9                    
                                                                                                          
Backtrace: 0x4008f6c3:0x3ffbb0e0 0x40082821:0x3ffbb100 0x400e6a95:0x3ffbb120 0x400e7132:0x3ffbb150 0x400d2
b0b:0x3ffbb180 0x400d0f2e:0x3ffbb360                                                                      
0x4008f6c3: is_free at /home/rajan/esp/esp-idf/components/heap/multi_heap.c:377                           
 (inlined by) multi_heap_malloc_impl at /home/rajan/esp/esp-idf/components/heap/multi_heap.c:429          
                                                                                                          
0x40082821: heap_caps_malloc at /home/rajan/esp/esp-idf/components/heap/heap_caps.c:130                   
                                                                                                          
0x400e6a95: printProportionalChar at /home/rajan/esp/hello_world/components/tft/tft.c:1523                

0x400e7132: TFT_print at /home/rajan/esp/hello_world/components/tft/tft.c:2019 (discriminator 1)

0x400d2b0b: app_main at /home/rajan/esp/hello_world/main/hello_world_main.c:165

0x400d0f2e: main_task at /home/rajan/esp/esp-idf/components/esp32/cpu_start.c:470
```

and resets.

AsyncAwayt, 2018-08-17:
Not having this issue on my applications. What fonts are you using?

rpatel3001, 2018-08-26:
I had left the font as default. This issue was solved by manually calling TFT_setFont(DEFAULT_FONT, 0)

MichaelEFlip, 2021-03-12:
I have the same problems here when using a TTGO T-DISPLAY (CONFIG_TFT_PREDEFINED_DISPLAY_TYPE = 5).

Characters "p" and "g" in TFT_print cause immediate resets of the ESP32. Setting the default font as mentioned above solves the problem.