loboris/ESP32_TFT_library #52: TFT_print() problem after TFT_read_touch() call

chrdan, 2018-08-28:
With TOUCH_TYPE_XPT2046 activated after calling TFT_read_touch() the TFT_print() command shows the first character of the given string crippled, which means the left and right part of that single character are swapped (for example an "O" changes from "()" to ")("), and its foreground color also isn't set correctly.

To get that result insert

```
  TFT_print("O_Untouched", CENTER, 80);
  vTaskDelay(10000 / portTICK_RATE_MS);
```
in tft_demo.c after line 996:

```
  if (TFT_read_touch(&tx, &ty, 0)) {
  ...
  }
  else {
    TFT_print("O_Untouched", CENTER, 80); // added
    vTaskDelay(10000 / portTICK_RATE_MS); // added
    doexit++;
```

Is there a display reinitialization problem after switching devices (disp_spi -> ts_spi -> disp_spi)?

Nice library, btw!

TIA

Chris

viethoangitvn, 2018-11-13:
Hi Chris,
    I am using ILI9341 and an NFC Reader that share the same SPI bus of ESP ( different CS pins)  and can see same problem. Did you find the reason of this issue?

Thank you, 
Viet Hoang.

sieren, 2019-04-18:
@chrdan do you mind posting your touch rewrites somewhere? I'm running into the same issue apparently.

mekkoli, 2021-07-19:
same problem for me with XPT2046 and loboris great library.
anyone has solved this issue ?
@chrdan have you posted anywhere your solution ?