loboris/ESP32_TFT_library #54: Lolin/Wemos 128x128 7735 almost works... but not quite :-(

ruffle-b, 2018-09-04:
I'm using a Wemos/Lolin 1.44" 128x128 7735 TFT.  It's the one that plugs into the Lolin D32 Pro and I think the same as the Adafruit 1.44" TFT.

I've set the display type to  DISP_TYPE_ST7735R (B seems the same) and DEFAULT_TFT_DISPLAY_WIDTH and DEFAULT_TFT_DISPLAY_HEIGHT to 128 and the tft_demo is sooooo close to working well.

What's not working is a sort of wrap/tear of the boxes at the top and bottom of the demo and the whole display being offset a few pixels up and to the left:
[image]
[image]

FWIW, I've tried two displays and two ESP32 boards. The hardware works with the Adafruit ST7735 Library in the Arduino IDE using tft.initR(INITR_144GREENTAB) but I'm trying to use this library with esp-idf.

I've tried playing with the XSTART and XEND settings (just flailing around in the dark really) but that doesn't seem to make any difference.

Can anyone point me in the right direction?

ruffle-b, 2018-09-05:
Oh dear.... it gets worse :-(

If you uncomment the test_sd_card() function in the demo and call it just before tft_demo:

`test_sd_card();
        tft_demo();`

it a) seems to corrupt the SDCard and b) the demo doesn't display anything else on the TFT after the "SPIFFS Mounted" message.

My ESP32 (Lolin D32 Pro) has the inbuilt SDCard and TFT with different CS pins on the same SPI bus.

Does this mean this library won't work with an SDCard on the same bus?

PS - I noticed 80000000 hard coded in spi_master_lobo.c and thought I might be on to something (I run my ESPS32 at 240MHz). Sadly changing the frequency of my ESP32 to 80MHz doesn't fix this problem.

sukeshak, 2018-09-24:
Yes this library has an issue where display and SD Card doesn't work together. I had to abandon this library for the same reason.

@loboris is aware of the issue but I guess busy to fix it ☹

nospam2000, 2018-12-02:
The offset needs to be changed for this display to x=2, y=3. You can use the original library and this derived class to fix it:

```
class WEMOS_TFT_144_V100 : public Adafruit_ST7735 {
  public:
    WEMOS_TFT_144_V100(int8_t _CS, int8_t _DC, int8_t _MOSI, int8_t _SCLK, int8_t _RST = -1, int8_t _MISO = -1) :
      //Adafruit_ST7735(_CS, _DC, _MOSI, _SCLK, _RST, _MISO) // _MISO is missing in this constructor, although it exists in the superclass
      Adafruit_ST7735(_CS, _DC, _MOSI, _SCLK, _RST)
    {
    }
    WEMOS_TFT_144_V100(int8_t CS, int8_t RS, int8_t RST = -1) :
      Adafruit_ST7735(CS, RS, RST)
    {
    }
    WEMOS_TFT_144_V100(SPIClass *spiClass, int8_t CS, int8_t RS, int8_t RST = -1) :
      Adafruit_ST7735(spiClass, CS, RS, RST)
    {
    }

    void initR(uint8_t /* options */, int8_t colstart = 2, int8_t rowstart = 3)
    {
      Adafruit_ST7735::initR(INITR_144GREENTAB);
      _colstart = colstart;
      _rowstart = rowstart;
      setRotation(0); // to set _xstart and _ystart
    }
};
```

create the tft instance like this:
`WEMOS_TFT_144_V100 tft = WEMOS_TFT_144_V100(TFT_CS, TFT_DC, TFT_RST);
`