loboris/ESP32_TFT_library #58: Adding support for CMake

Grtschnk, 2018-10-05:
Since ESP-IDF 3.1 Espressif included CMake support and is making it the default compiler.

Make projects will still be supported, but as far as I understand it, I cannot use this great TFT library with a CMake project because the "components" are missing the CMakeLists.txt. Espressif added these in the v3.1 release for all components.

In short: CMakeLists.txt are needed for the "components" in said folder.

Grtschnk, 2018-10-05:
I am currently trying to wrap my head around the new CMake toolchain and CMake in general.
I guess there must be a way to easily generate these files (or reverse-engineer them from the makefiles).

Maybe Espressif itself would be interesting in helping to upgrade your library? I saw they included an external MQTT in their master now.

Grtschnk, 2018-10-07:
Think I understand the new CMake toolchain: I created the missing CMakeLists.txt, however I still have problems with the linker. Not sure if this is an error of the ESP-IDF when compiling c (not cpp) custom components or of my IDE-Plugin (VisualGDB). I opened a ticket, let's see.

Grtschnk, 2018-10-08:
I created my CMake files locally and it works fine for me. If you like I can send them to you and/or you enable push for the repo.

arnold-b, 2018-12-15:
Hello,

could you please help. how can i use this library with cmake. Could i get your projekt.  Thanks

Grtschnk, 2018-12-16:
These are my local CMakeFiles. Espressif did a great job creating an easy-to-use toolchain. I suggest you have a look at their examples (and the ones below) and create the files yourself to exactly suit your project.

`set(COMPONENT_SRCS tft.c tftspi.c comic24.c def_small.c DefaultFont.c DejaVuSans18.c DejaVuSans24.c minya24.c SmallFont.c tooney32.c Ubuntu16.c)
set(COMPONENT_ADD_INCLUDEDIRS "." "../spidriver")

register_component()`

`set(COMPONENT_SRCS spi_master_lobo.c)
set(COMPONENT_ADD_INCLUDEDIRS ".")

register_component()`

Cheers

arnold-b, 2018-12-28:
Hi,

 

thanks for your make file. 

 

I got I already running with esp-idf .30 

 

But I still have problem to use 3.3 ,

 

I just bought a wrover kit and I wanted to test the TFT.

This library is realy good.  I will use it to create a small GUI. 

Currently I´m fighting a bit with SMTPE 610.

 

Before I manly used Arduino Ide, but this one is too slow. I could not even make I2C protocol analyser for 400khz. 

Now  I´m using Visual Studio and testing VisualGDB. 

 

Thanks Arnold

rangeli, 2019-02-08:
I've added CMake support in my fork.  See pull request above.