loboris/ESP32_TFT_library #65: TFT Display flipped/mirrored on ESP-WROVER-KIT V4.1

DenisCrnic, 2019-01-06:
After downloading and compiling your library on ESP-WROVER-KIT V4.1 display runs the demo program but the screen is flipped.
I've used the following configuration:
```
#define DEFAULT_DISP_TYPE DISP_TYPE_ILI9341
#define DEFAULT_TFT_DISPLAY_WIDTH   240
#define DEFAULT_TFT_DISPLAY_HEIGHT  320
#define DISP_COLOR_BITS_24          0x66
#define DEFAULT_GAMMA_CURVE         0
#define DEFAULT_SPI_CLOCK           26000000
#define TFT_INVERT_ROTATION         0
#define TFT_INVERT_ROTATION1        1
#define TFT_INVERT_ROTATION2        0
#define TFT_RGB_BGR                 0x08
```

```
spi_lobo_bus_config_t buscfg={
        .miso_io_num=25,                // set SPI MISO pin
        .mosi_io_num=23,                // set SPI MOSI pin
        .sclk_io_num=19,                // set SPI CLK pin
        .quadwp_io_num=-1,
        .quadhd_io_num=-1,
        .max_transfer_sz = 6*1024,
    };
spi_lobo_device_interface_config_t devcfg={
        .clock_speed_hz=26000000,                // Initial clock out at 8 MHz
        .mode=0,                                // SPI mode 0
        .spics_io_num=-1,                       // we will use external CS pin
        .spics_ext_io_num=22,           // external CS pin
        .flags=LB_SPI_DEVICE_HALFDUPLEX,        // ALWAYS SET  to HALF DUPLEX MODE!! for display spi
};
```

[image]
[image]

curiousmuch, 2019-01-12:
```c
#define DEFAULT_DISP_TYPE           DISP_TYPE_ILI9341
#define DEFAULT_TFT_DISPLAY_WIDTH   240
#define DEFAULT_TFT_DISPLAY_HEIGHT  320
#define DISP_COLOR_BITS_24          0x66
#define DEFAULT_GAMMA_CURVE         0
#define DEFAULT_SPI_CLOCK           26000000
#define TFT_INVERT_ROTATION         0
#define TFT_INVERT_ROTATION1        0
#define TFT_INVERT_ROTATION2		0
#define TFT_RGB_BGR                 0x00
```
A little experimentation.

PaladinEng, 2019-04-29:
I'm having this issue as well. 
I'm using this display:
[this dsiplay](https://www.buydisplay.com/default/1-44-inch-tft-touch-screen-serial-spi-128x128-lcd-display-module-st7735) which includes an st7735S controller. 

Any ideas how to fix this?

PaladinEng, 2019-04-29:
Update: fixed the mirroring problem with the following settings: 
#define TFT_INVERT_ROTATION 0
#define TFT_INVERT_ROTATION1 1

exislow, 2019-06-23:
I fixed it and added a new menuconfig option for ESP-WROVER-KIT 4.1. Pull request: https://github.com/loboris/ESP32_TFT_library/pull/71