loboris/ESP32_TFT_library #66: How to configure SPIFFS

pavvank, 2019-01-17:
Hey i am not able to configure the spiffs. As my screen shows the colour but there is  no text instead a black rectangle is there(at place of text).

zoon81, 2019-07-10:
Hi, the `make copyfs` command not worked for me. I used the next command to get things work after I compiled the spiffs_image folder.
`esptool.py --chip esp32 --port /dev/cu.SLAB_USBtoUART --baud 921600 --before default_reset --after hard_reset write_flash -z 0x180000 ./components/spiffs_image/spiffs_image.img`

jeremyjh, 2019-09-17:
On more recent versions of esp-idf ESPPORT variable won't be set by Kconfig, its only ever passed as an arg to idf.py. A workaround that works on current esp-idf master is `ESPPORT=/dev/cu.SLAB_USBtoUART make copyfs`. Of course, your port may be different. You don't have to compile the image yourself unless you want to change it.

pantalok, 2020-03-07:
Hi, I am new to ESP32 so may be its my lack of info but I see the below issue. 

When I try to flash the SPIFFS image ESP32 bootloader fails. Any idea what I may be missing
Thanks

I am using ESP32 wrover kit. I am able to get this library working (great library! BTW) after I apply these two changes
1. https://github.com/zoon81/ESP32_TFT_library/commit/0ee6fd16d20f2a52e6ae2fa769485205a4afd69f
2. https://www.bountysource.com/issues/57372755-latest-esp-wrover-kit-uses-ili9341
and chage the SPI flash size from 2MB to 4MB under menuconfig

It i perform regular flash update (i.e. make flash) lash it all works great I can see all these test screens

Next I attempted to show jpeg/bmp
However whenever I perform make copyfs (or flash fs) to flash a SPIFS I see following exceptions and failures 

```
rst:0x3 (SW_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)^M
configsip: 0, SPIWP:0xee^M
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00^M
mode:DIO, clock div:2^M
load:0x3fff0030,len:4^M
load:0x3fff0034,len:7148^M
load:0x40078000,len:14996^M
load:0x40080400,len:4432^M
entry 0x400806e8^M
^[[0;32mI (56) boot: ESP-IDF v4.2-dev-459-ge36516372-dirty 2nd stage bootloader^[[0m^M
^[[0;32mI (56) boot: compile time 08:34:36^[[0m^M
^[[0;32mI (58) boot: chip revision: 1^[[0m^M
^[[0;32mI (62) boot_comm: chip revision: 1, min. bootloader chip revision: 0^[[0m^M
^[[0;32mI (76) boot.esp32: SPI Speed      : 40MHz^[[0m^M
^[[0;32mI (76) boot.esp32: SPI Mode       : DIO^[[0m^M
^[[0;32mI (78) boot.esp32: SPI Flash Size : 4MB^[[0m^M
^[[0;32mI (83) boot: Enabling RNG early entropy source...^[[0m^M
^[[0;32mI (88) boot: Partition Table:^[[0m^M
^[[0;32mI (92) boot: ## Label            Usage          Type ST Offset   Length^[[0m^M
^[[0;32mI (99) boot:  0 nvs              WiFi data        01 02 00009000 00006000^[[0m^M
^[[0;32mI (106) boot:  1 phy_init         RF data          01 01 0000f000 00001000^[[0m^M
^[[0;32mI (114) boot:  2 factory          factory app      00 00 00010000 00100000^[[0m^M
^[[0;32mI (122) boot:  3 storage          Unknown data     01 82 00180000 00100000^[[0m^M
^[[0;32mI (129) boot: End of partition table^[[0m^M
^[[0;32mI (133) boot_comm: chip revision: 1, min. application chip revision: 0^[[0m^M
^[[0;32mI (141) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0ab58 ( 43864) map^[[0m^M
^[[0;32mI (165) esp_image: segment 1: paddr=0x0001ab80 vaddr=0x3ffb0000 size=0x050f4 ( 20724) load^[[0m^M
^[[0;32mI (174) esp_image: segment 2: paddr=0x0001fc7c vaddr=0x40080000 size=0x0039c (   924) load^[[0m^M
^[[0;32mI (175) esp_image: segment 3: paddr=0x00020020 vaddr=0x400d0020 size=0x24f40 (151360) map^[[0m^M
^[[0;31mE (235) esp_image: invalid segment length 0xeb7596c0^[[0m^M
**^[[0;31mE (235) boot: Factory app partition is not bootable^[[0m^M
^[[0;31mE (235) boot: No bootable app partitions in the partition table^[[0m^M**
```