loboris/ESP32_TFT_library #73: error: self-comparison always evaluates to false

hcmila, 2019-07-07:
Toolchain path: .../esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp32-2019r1
Compiler version: 8.2.0

.../ESP32_TFT_library/components/spiffs/spiffs_nucleus.c:682:346:
error: self-comparison always evaluates to false [-Werror=tautological-compare]
   SPIFFS_VALIDATE_OBJIX(objix->p_hdr, obj_id, objix->p_hdr.span_ix);

earosenfeld, 2019-07-09:
I am getting the same error

zoon81, 2019-07-10:
Hi,
I compiled yesterday, check my fork or check [this](https://github.com/zoon81/ESP32_TFT_library/commit/0ee6fd16d20f2a52e6ae2fa769485205a4afd69f) commit.
Some changes are made on ESP-IDF since this lib are published. This is why we get errors during compiling the demo.

earosenfeld, 2019-07-10:
Thank you for the commit, this fixed my issue. Your help is much appreciated.

AlexiLlaiho, 2019-08-06:
Dear, zoon81! Thank you so much! You've saved my day:)