loboris/ESP32_TFT_library #78: How to use external fonts in code?

MusabAybek, 2019-09-11:
Hi. I'm new on idf coding and got an issue about fonts. I created my own .c file font via ttf2c_vc2003.exe program. But I don't know how to use this external font in my code. Can somebody explain please? 
  Thanks in advance...

Mscichu, 2019-09-13:
Hi!

First use program: https://github.com/loboris/ESP32_TFT_library/tree/master/tools
	ttf2c_vc2003.exe 100 Zebulon.otf zebi.c 
and edit as in readme.txt
Next you must write file to filesystem and convert .c file to .fon in program using command: 
	compile_font_file("/spiffs/fonts/zebi.c",1); // zebi.c for example

at last you can use new font:
	TFT_setFont(USER_FONT, "/spiffs/fonts/zebi.fon");

MusabAybek, 2019-09-14:
Thank you so much for your helpful reply...
I converted my .ttf file to .c and edited as in readme.txt
But the command you provided didn't work. I located my terminal to the destination where my .c type font file is exist (C:\msys32\home\Musab\esp\workspace\ESP32_TFT_library-master\components\tft)
then I wrote; compile_font_file("/spiffs/fonts/oxygen.c",1); into terminal but didn't work, I got error says  " 'compile_font_file' is not recognized as an internal "
Do I need to write my .c type font file location instead of "/spiffs/fonts/"?

Mscichu, 2019-09-14:
You must copy oxygen.c to ESP32_TFT_library\components\spiffs_image\image\fonts folder, then in command line use:
make makefs
and:
make flashfs

to write new filesystem data to ESP

MusabAybek, 2019-09-14:
After I copy .c file into \spiffs_image\image\fonts , is file will be automatically converted to the .fon type file? Because I'm stuck on converting from .c to .fon

Mscichu, 2019-09-14:
No, from this folder you copy files to ESP using flahfs command.  Next you must use compile_font_file in your program to read .c file from ESP and make .fon file. Then you can use TFT_setFont(USER_FONT, "/spiffs/fonts/oxygen.fon");

MusabAybek, 2019-09-15:
Thank you for your really helpful replies. I got so far with your help. I did everything you wrote, everything on the code side looks okay, But I got still some issues.. 
I attached a part from my example code here, I'm trying to show different kinds of fonts one by one. But still see the same font every time sentences appear on the TFT.
[example.txt](https://github.com/loboris/ESP32_TFT_library/files/3614098/example.txt)

Mscichu, 2019-09-15:
Ok. Paste what debug shows

MusabAybek, 2019-09-15:
I'm really beginner on Eclipse and IDF coding, sorry I don't understand clearly. Is that what you wanted me to send? Copied from Debugger Console. 
[example1.txt](https://github.com/loboris/ESP32_TFT_library/files/3614124/example1.txt)

MusabAybek, 2019-09-15:
When I enter "make monitor", it says;
 Error opening source file '/spiffs/fonts/bignoodle.c'
 Error opening source file '/spiffs/fonts/codebold.c'
 Error opening source file '/spiffs/fonts/heavitas.c'
 Error opening source file '/spiffs/fonts/moonheavy.c'
 Error opening source file '/spiffs/fonts/professor.c'

What is the reason could be?

Mscichu, 2019-09-15:
You are not doing correctly make makefs and make flashfs so you don't have file in ESP.

MusabAybek, 2019-09-15:
This is the output of terminal, I tried to repeat all the process;
 
`Musab@MUSAB-PC MINGW32 ~
$ cd ~/esp/workspace/ESP32_TFT_library-master

Musab@MUSAB-PC MINGW32 ~/esp/workspace/ESP32_TFT_library-master
$ make makefs
/home/Musab/esp/workspace/ESP32_TFT_library-master/components/mkspiffs/Makefile.projbuild:13: warning: overriding recipe for target 'clean'
/home/Musab/esp/esp-idf/components/app_update/Makefile.projbuild:48: warning: ignoring old recipe for target 'clean'
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Makefile:65: warning: undefined variable 'TARGET_CXXFLAGS'
Building mkspiffs ...
Making spiffs image ...
python /home/Musab/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port COM5 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect
/.
/fonts/.
/fonts/BigFont.fon
/fonts/bignoodle.c
/fonts/codebold.c
/fonts/DejaVuSans12.fon
/fonts/DejaVuSans18.fon
/fonts/DejaVuSans24.fon
/fonts/DotMatrix_M.fon
/fonts/Grotesk24x48.fon
/fonts/heavitas.c
/fonts/moonheavy.c
/fonts/ocrfont.c
/fonts/oxygen.c
/fonts/professor.c
/fonts/swiss721_outline.fon
/images/.
/spiffs.info

Musab@MUSAB-PC MINGW32 ~/esp/workspace/ESP32_TFT_library-master
$ make flashfs
/home/Musab/esp/workspace/ESP32_TFT_library-master/components/mkspiffs/Makefile.projbuild:13: warning: overriding recipe for target 'clean'
/home/Musab/esp/esp-idf/components/app_update/Makefile.projbuild:48: warning: ignoring old recipe for target 'clean'
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Making spiffs image ...
python /home/Musab/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port COM5 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect
/.
/fonts/.
/fonts/BigFont.fon
/fonts/bignoodle.c
/fonts/codebold.c
/fonts/DejaVuSans12.fon
/fonts/DejaVuSans18.fon
/fonts/DejaVuSans24.fon
/fonts/DotMatrix_M.fon
/fonts/Grotesk24x48.fon
/fonts/heavitas.c
/fonts/moonheavy.c
/fonts/ocrfont.c
/fonts/oxygen.c
/fonts/professor.c
/fonts/swiss721_outline.fon
/images/.
/spiffs.info
esptool.py v2.8-dev
Serial port COM5
Connecting........_____....._____....._____.
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:71:bf:52:dc:98
Uploading stub...
[... 149 lines cut ...]
I (3103) [SPIFFS]:   Work buffer: 2048 B
I (3108) [SPIFFS]:    FDS buffer: 384 B
I (3112) [SPIFFS]:    Cache size: 2048 B
I (3173) [SPIFFS]: Mounted

==========================================
Display: ILI9341: PORTRAIT 240,320 Color

Error opening source file '/spiffs/fonts/bignoodle.c'
Error opening source file '/spiffs/fonts/codebold.c'
Error opening source file '/spiffs/fonts/heavitas.c'
Error opening source file '/spiffs/fonts/moonheavy.c'
Error opening source file '/spiffs/fonts/professor.c'
font demo bitti

==========================================
Display: ILI9341: LANDSCAPE 320,240 Color

Error opening source file '/spiffs/fonts/bignoodle.c'
Error opening source file '/spiffs/fonts/codebold.c'
Error opening source file '/spiffs/fonts/heavitas.c'
Error opening source file '/spiffs/fonts/moonheavy.c'
Error opening source file '/spiffs/fonts/professor.c'
font demo bitti
`

Mscichu, 2019-09-15:
Hmmm... I don't know why it does't work. Try run standard fonts from flash, example:

	TFT_setFont(USER_FONT, "/spiffs/fonts/BigFont.fon");

MusabAybek, 2019-09-16:
Okay, thank you so much for your help, even so I'll keep trying. 
There is a code in the example code;
static const char *file_fonts[3] = {"/spiffs/fonts/DotMatrix_M.fon", "/spiffs/fonts/Ubuntu.fon", "/spiffs/fonts/Grotesk24x48.fon"};

Do you think it's something important we need to consider?

MusabAybek, 2019-09-17:
(Update) Changed to new microcontroller, still getting the same error...
Error opening source file '/spiffs/fonts/bignoodle.c'
Error opening source file '/spiffs/fonts/codebold.c'
Error opening source file '/spiffs/fonts/heavitas.c'
Error opening source file '/spiffs/fonts/moonheavy.c'
Error opening source file '/spiffs/fonts/professor.c'

MusabAybek, 2019-09-17:
compile_font_file( );  function does nothing. It's code is missing in tft.h library...

jeremyjh, 2019-09-17:
Does the test demo work if you just do `make copyfs` (uses image binary in this repo) rather than build the image yourself?

Mscichu, 2019-09-17:
I have another solution.
1. In tft.h around line 130 define your font name example:
#define OXYGEN_FONT 11 
2. In tft.c around line 1515 add another line:
else if (font == OXYGEN_FONT) cfont.font = oxygen32;
where oxygen32 is name your table in oxygen.c file
3. Add prepared oxygen.c file to components/tft

Now you don't need to write filesystem to ESP flash, just use make flash

MusabAybek, 2019-09-17:
> I have another solution.
> 
> 1. In tft.h around line 130 define your font name example:
>    #define OXYGEN_FONT 11
> 2. In tft.c around line 1515 add another line:
>    else if (font == OXYGEN_FONT) cfont.font = oxygen32;
>    where oxygen32 is name your table in oxygen.c file
> 3. Add prepared oxygen.c file to components/tft
> 4. Use TFT_setFont(OXYGEN_FONT, NULL); in project
> 
> Now you don't need to write filesystem to ESP flash, just use make flash

Thank you really much man... This solution works... 
I had to also add definitions for external fonts in tft.c file, around line 45
extern uint8_t oxygen32[]; //for example