loboris/ESP32_TFT_library #82: Printing variables with TFT_print function?

MusabAybek, 2019-11-21:
Hi everyone. I want to print varibles with TFT_print function.  When I try to write variable into function, I get error; "CPU halted."

int x = 10;
TFT_print( x ,10 ,10);

Isn't it possible to use this function like this? If not, how to write varibles onto screen with TFT_print function?
Thanks in advance...

elieDaan, 2019-11-22:
Hey, you have to convert your int into a char.

MusabAybek, 2019-11-22:
I tried but getting the same error :/

jeremyjh, 2019-11-22:
print takes a char * (string) argument.

MusabAybek, 2019-11-22:
What I want to do is something like that;

int x=0;
while(1)  {
x++;
TFT_print(x,10.10);
vTaskDelay(1000/portTICK_PERIOD_MS);   }

It didn't work when I try to write TFT_print(char *x,10,10);
Thanks for the answers, but I couldn't make it work :/

jeremyjh, 2019-11-22:
That is not how you convert an int to a string in C. The starting point here is the stdlib function `sprintf`. But you will first have to allocate enough memory for your string. Here is an example where they allocate some memory, declare some ints, and then make a string with those ints. 

https://www.geeksforgeeks.org/sprintf-in-c/

MusabAybek, 2019-11-23:
Thank you so much guys... It's working

closed 2019-11-24 by MusabAybek