loboris/ESP32_TFT_library #93: ST7789V2 support on M5StickC-Plus

aknabi, 2021-03-22:
Trying to get ST7789V2 support for the M5StickC-Plus running (new version of the M5StickC so assume it'll be a popular device)... backlighting on/off works, but display still doesn't show anything.

Of course assuming ST7789V will work for the ST7789V2 (will be checking the data sheets just in case).

I'm debugging this so if I find a solution will submit it.. in the meantime if anyone has any pointers, feedback would of course be appreciated.

added to CONFIG_EXAMPLE_DISPLAY_TYPE options in tftspi.h:

`#elif CONFIG_EXAMPLE_DISPLAY_TYPE == 5

// ** Set the correct configuration for M5StickC Plus TFT
// ---------------------------------------------------------
#define DEFAULT_DISP_TYPE           DISP_TYPE_ST7789V
#define DEFAULT_TFT_DISPLAY_WIDTH   135
#define DEFAULT_TFT_DISPLAY_HEIGHT  240
#define DISP_COLOR_BITS_24          0x66
#define DEFAULT_GAMMA_CURVE         0
#define DEFAULT_SPI_CLOCK           26000000
#define TFT_INVERT_ROTATION         0
#define TFT_INVERT_ROTATION1        1
#define TFT_RGB_BGR                 0x00

#define USE_TOUCH                   TOUCH_TYPE_NONE

#define PIN_NUM_MISO -1		// SPI MISO
#define PIN_NUM_MOSI 15		// SPI MOSI
#define PIN_NUM_CLK  13		// SPI CLOCK pin
#define PIN_NUM_CS   5		// Display CS pin
#define PIN_NUM_DC   23		// Display command/data pin
#define PIN_NUM_TCS  0		// Touch screen CS pin (NOT used if USE_TOUCH=0)

#define PIN_NUM_RST  18  	// GPIO used for RESET control (#16)
#define PIN_NUM_BCKL 0  	// GPIO used for backlight control
#define PIN_BCKL_ON  1  	// GPIO value for backlight ON
#define PIN_BCKL_OFF 0  	// GPIO value for backlight OFF`

`df`

tuupola, 2021-03-22:
Wild guess is that `PIN_NUM_TCS` and `PIN_NUM_BCKL` should be `-1` since they are not used. Other than that pins look correct. M5StickC-Plus also is able to run with 40000000HZ clock. Below is my config from another library:

```
CONFIG_MIPI_DISPLAY_SPI_CLOCK_SPEED_HZ=40000000
CONFIG_MIPI_DISPLAY_PIN_MISO=-1
CONFIG_MIPI_DISPLAY_PIN_MOSI=15
CONFIG_MIPI_DISPLAY_PIN_CLK=13
CONFIG_MIPI_DISPLAY_PIN_CS=5
CONFIG_MIPI_DISPLAY_PIN_DC=23
CONFIG_MIPI_DISPLAY_PIN_RST=18
CONFIG_MIPI_DISPLAY_PIN_BL=-1
CONFIG_MIPI_DISPLAY_PWM_BL=-1
```

aknabi, 2021-03-23:
Thanks for the reply... some good pointers... note (at least in my code) the PIN_NUM_TCS is used (and the gpio config'd) only if USE_TOUCH != 0 (TOUCH_TYPE_NONE)... ensured that gpio code isn't called... on the PIN_NUM_BCKL that looks for the #define (#ifdef) so commented that out to skip the gpio config for BCKL (as on M5StickCs that's in the power chip). Still have the same behavior (trying tweaking some other settings on a lark). Next step for me will be to create a new project just isolating this code and getting it to run on the M5StickC-Plus... hopefully can come back with some positive results soon

tuupola, 2021-03-24:
You could also check if the AXP192 is initialised properly. It provides power both for the display driver chip and the backlight. Are you using ESP-IDF?

aknabi, 2021-03-25:
Yeah... I am... the backlight actually works (can turn it on and off)... Note one thing I found in another library was this in setup:

``
TFT_setclipwin(40, 52, 279, 186);
``

Added that to no avail... still working on creating a clean project with just the basic init and TFT code and going from there (will be a weekend project)... thanks again for the feedback

tuupola, 2021-03-26:
If you don't mind testing with another graphics library you could rule out possibility of broken display with [one of my apps](https://github.com/tuupola/esp_effects). Is just tested and it runs fine on two of my M5StickC Plus boards. If it also works for you the you could then compare the settings etc.

```
$ git clone git@github.com:tuupola/esp_effects.git --recursive
$ cd esp_effects
$ cp sdkconfig.m5stickc-plus sdkconfig
$ make -j8 flash
```

Or if you prefer using the new `idf.py` tool the something like:

```
$ git clone git@github.com:tuupola/esp_effects.git --recursive
$ cd esp_effects
$ cp sdkconfig.m5stickc-plus sdkconfig
$ idf.py build
$ idf.py flash -p /dev/cu.usbserial-095268E9A8 -b 115200
```

aknabi, 2021-03-26:
Thanks for that... installed the latest ESP-IDF and got that running... Just FYI I had an issue as follows when running:

``
E (4328) i2c: i2c_param_config(644): i2c clock choice is invalid, please check flag and frequency
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x40086c50
file: "./components/esp_i2c_helper/i2c_helper.c" line 57
func: i2c_init
expression: i2c_param_config(port, &conf)
``

which is mentioned in this issue: https://github.com/espressif/esp-idf/issues/6293

The solution mentioned of setting conf.clk_flags = 0; in the init worked
(I add this in components/esp_i2c_helper/i2c_helper.c right above 
ESP_ERROR_CHECK(i2c_param_config(port, &conf)); )

My M5StickC Plus is running your (cool) demo fine, so now I can use that to figure out what's up.

Thanks again