loboris/ESP32_TFT_library #95: ESP-IDF now using CMAKE code no longer works

daverathbone, 2021-04-26:
This code is no longer supported by ESP-IDF ver 4.xx as Espressif have moved to a CMAKE structure.
Is it possible you could update the build?

nhancaheo, 2021-05-01:
> This code is no longer supported by ESP-IDF ver 4.xx as Espressif have moved to a CMAKE structure.

Same issue here
`CC build/heap/heap_caps_init.o
/home/dolphin/esp/esp-idf/components/heap/heap_caps_init.c: In function 'heap_caps_add_region_with_caps':
/home/dolphin/esp/esp-idf/components/heap/heap_caps_init.c:230:44: error: the comparison will always evaluate as 'true' for the address of 'registered_heaps_write_lock' will never be NULL [-Werror=address]
     MULTI_HEAP_LOCK(&registered_heaps_write_lock);
                                            ^~
/home/dolphin/esp/esp-idf/components/heap/heap_caps_init.c:232:45: error: the comparison will always evaluate as 'true' for the address of 'registered_heaps_write_lock' will never be NULL [-Werror=address]
     MULTI_HEAP_UNLOCK(&registered_heaps_write_lock);
                                             ^~
cc1: some warnings being treated as errors
/home/dolphin/esp/esp-idf/make/component_wrapper.mk:291: recipe for target 'heap_caps_init.o' failed
make[1]: *** [heap_caps_init.o] Error 1
/home/dolphin/esp/esp-idf/make/project.mk:642: recipe for target 'component-heap-build' failed
make: *** [component-heap-build] Error 2
`

sineer, 2021-11-01:
This fork works!

https://github.com/jeremyjh/ESP32_TFT_library

Working on my TTGO-T1 display out-of-the-box w/ IDF 4.3.0 and platformio...

Thanks @jeremijh